DiSlord/NanoVNA-D #1: Make fails, mising nano.specs file

galileo-pkm, 2020-09-21:
Make fails with:
arm-none-eabi-gcc: fatal error: cannot read spec file 'nano.specs': No such file or directory

Uncommitted file?

DiSlord, 2020-09-24:
no 'nano.specs' file in my NanoVNA folder
See
https://stackoverflow.com/questions/26931979/gnu-arm-nano-specs-not-found

galileo-pkm, 2020-09-24:
Installing arm-none-eabi-newlib package (Fedora 32) fixes that error.  
The nano in the file name threw me of ...

Unfortunately compile now fails with a nonsensical error: 

`Compiling hal_lld.c
ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c:54:40: error: expected identifier or '(' before 'void'
   54 | static void hal_lld_backup_domain_init(void) {
      |                                        ^~~~
ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c: In function 'hal_lld_init':
ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c:235:30: error: expected expression before ')' token
  235 |   hal_lld_backup_domain_init();
      |                              ^
make: *** [ChibiOS/os/common/startup/ARMCMx/compilers/GCC/rules.mk:216: build/obj/hal_lld.o] Error 1
`

Kind of strange considering that edy555 repo compiles without issues.

DiSlord, 2020-09-25:
Comment call (it not used, and should be disabled)
// hal_lld_backup_domain_init();
in "**ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c**"

Also need change in "**ChibiOS\os\hal\ports\STM32\LLD\ADCv3\hal_adc_lld.c**"
OSAL_IRQ_HANDLER(STM32_ADC1_HANDLER) {
to
**__attribute__((weak))** OSAL_IRQ_HANDLER(STM32_ADC1_HANDLER) {

igor-m, 2020-09-25:
`__attribute__((weak)) OSAL_IRQ_HANDLER(STM32_ADC1_HANDLER) { `

DiSlord, 2020-09-25:
Yes, __ lost on formatting

DiSlord, 2020-09-25:
__dfsfs__

galileo-pkm, 2020-09-25:
If you are modifying upstream libraries you should fork ChibiOS 
or add comments about the changes in the Readme. 
For now I have just commented out the -DVNA_AUTO_SELECT_RTC_SOURCE line in the Makefile.
It builds, haven't tested it yet.

DiSlord, 2020-09-26:
On commented VNA_AUTO_SELECT_RTC_SOURCE you need manually select RTC source, by default use LSI

galileo-pkm, 2020-09-26:
Ok, thanks

closed 2020-11-17 by galileo-pkm