DiSlord/NanoVNA-D #2: Add version FW for default NanoVNA-H

ruzanow, 2020-10-16:
Now the menu contains useless items about (non-existent) SD card. The clock is also semi-working.
Or pls, add some options in console to hide that.

DiSlord, 2020-10-17:
You can compile firmware for self and disable SD Card and/or clock option
See nanovna.h file:
```
// Add RTC clock support
#define __USE_RTC__
// Add SD card support, req enable RTC (additional settings for file system see FatFS lib ffconf.h)
#define __USE_SD_CARD__

```
just comment this strings

you can soldier SD card slot in device (and external quartz rot clock) see video https://www.youtube.com/watch?v=173gnsfrCbQ
External 32768 Hz quartz can be connected to PC14 and PC15 pins, it allow run clock while device power off.

ruzanow, 2020-10-17:
Ok, thanks. I can disable SD and make FW. 
However, there is an error when RTC is disabled:

```
ui.c: In function 'show_version':
ui.c:545:17: error: 'buffer' undeclared (first use in this function)
     plot_printf(buffer, sizeof(buffer), "Batt: %d.%03dV", vbat/1000, vbat%1000);
```

Also, it's good to update links in README.md to
```
$ git clone https://github.com/DiSlord/NanoVNA-D.git
$ cd NanoVNA-D
```

DiSlord, 2020-10-17:
I know about this, fix this and some others on next commit
(Need moe buffer declaration outside RTC)
```
+    char buffer[32];
#ifdef __USE_RTC__
-    char buffer[32];
```

ruzanow, 2020-10-22:
Don't work. I build the firmware and write it. NanoVNA now freezes after setting the frequency. 
[NanoVNA H v1.0.39.zip](https://github.com/DiSlord/NanoVNA-D/files/5422072/NanoVNA.H.v1.0.39.zip)

ruzanow, 2020-10-22:
BTW, default code for NanoVNA-H or NanoVNA-H4?

DiSlord, 2020-10-22:
Default is nanoVNA-H, for H4 need switch to H4 branch
Yes, i made a lot of code size fixes/optimisations. You can also disable serial code support (allow connect to CPU over Bluetooth/WiFi modules)
Just disable in nanovna.h
`#define __USE_SERIAL_CONSOLE__`
In this case code size should be ~77k (i compile last git vs disabled
[NanoVNA H 1.0.40 no (SD Serial RTC).zip](https://github.com/DiSlord/NanoVNA-D/files/5424564/NanoVNA.H.1.0.40.no.SD.Serial.RTC.zip)

ruzanow, 2020-10-22:
Ok, finally it's worked (your and my build). Thanks.

closed 2020-10-22 by ruzanow