DiSlord/NanoVNA-D #4: Feature request: disable screen

galileo-pkm, 2020-11-17:
Is it possible to implement a serial command to disable the screen?
I would like to test the screens impact on overall device performance
while avoiding the hassle of physically removing the screen from the
device. 
Also it would be beneficial to reduce the device power consumption
while connected to a mobile device.

nlroth, 2020-11-17:
The only way to trully disable the screen is to cut the power going to it.
The uP on the display will always be running and generating a certain level of RFI even if the LCD panel is not refreshed.
Now, with that said, PA15 controls the reset line to the panel - but I'm not sure if holding the reset active will reduce the noise level.
As it is right now, the LCD scanning is suspended during an RF sweep to reduce noise.

roseengineering, 2023-10-26:
The screen on my Nanovna-H broke a while back.  However it still worked.  I could still control it using the serial port.    Headless measurement operation was possible.  Unfortunately the latest 2023 version of the Nanovna-D firmware does not function without a screen: no blinking lights, no serial port access.  However version 1.0.38 does work without a screen, and you can control the device using the serial port.  I haven't checked firmware versions in between.

closed 2024-02-04 by DiSlord