DiSlord/NanoVNA-D #5: Scannning start and stop frequencies inserted into the file names (files on the sdcard)

igor-m, 2020-12-28:
While messing with a multiband antenna I have to measure at several frequency bands.. After saving the s1p files to the sdcard (H4) I always have to open the files in order to see the start/stop scanning frequencies, a pity :)
Thus I've created a quick hack -  inserting the start/stop freq information into the filename. 
These are the few changes I've done:

file ffconf.h line 100
#define FF_MAX_LFN  64

file ffconf.h line 131
#define FF_LFN_BUF  64

file ui.c line 1128
plot_printf(fs_filename, FF_LFN_BUF, "VNA_%06X_%06X_%uk_%uk.s%dp", dr, tr, frequencies[0]/1000, frequencies[sweep_points-1]/1000,data);

Now I get the filenames stored on the sdcard with start/stop in kHz:
[image]

You have to adjust the size of the "saving information" window on the LCD as well (I did not yet).

nlroth, 2020-12-29:
Nice hack - thanks!

DiSlord, 2022-05-30:
Added filename input i think solve this issue

closed 2022-05-30 by DiSlord