DiSlord/NanoVNA-D #6: time command appears to be broken after vers 1.0.45

nlroth, 2021-03-05:
Can set time on 1.0.45 but on 1.0.48, it no longer works:

ch> time b 0x210305 0x083322
ch> time
2021/03/05 09:45:29
usage: time {[y|m|d|h|min|sec] 0-99} or {b 0xYYMMDD 0xHHMMSS}
ch>

DiSlord, 2021-03-05:
In some cases time command not run from one call (need more timeout??)

Try launch this command several time

nlroth, 2021-03-05:
I already tried the command several times manually through the console and using OneofEleven's Nanovna-App time-sync in the setup.
It works without issue on your 1.0.46 F/W but I had issues compiling your 1.0.48
Will commenting-out    UDEFS+ = -DVNA_AUTO_SELECT_RTC_SOURCE     in makefile, disable time set ability?

DiSlord, 2021-03-05:
Last time i change RTC related code in 1.0.43 (Increase RTC ready time wait (fix time set in some cases))

**Time tick on version screen?**

> Will commenting-out    UDEFS+ = -DVNA_AUTO_SELECT_RTC_SOURCE     in makefile, disable time set ability?
If comment this - not need change ChibiOS sources for disable hal_lld_backup_domain_init function
But need set manually UDEFS+ = -DVNA_USE_LSE for LSE use (or LSI used by default)

Also you can check console commands
'time y 21'
'time m 3'
....
'time sec 0'

nlroth, 2021-03-09:
I followed info from issue #1 but I still had issues. I commented out the lines in the makefile but the console time command still did not work. I then discovered that if the space after the 1st void is removed in line 54 in ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c :

change:    static void hal_lld_backup_domain_init(void) {
   to          static voidhal_lld_backup_domain_init(void) {
 and also comment out:
// hal_lld_backup_domain_init();

I now get this compile error But the code compiles:
        'voidhal_lld_backup_domain_init' defined but not used

Also need to change in "ChibiOS\os\hal\ports\STM32\LLD\ADCv3\hal_adc_lld.c"
      OSAL_IRQ_HANDLER(STM32_ADC1_HANDLER) {
to    __attribute((weak))__ OSAL_IRQ_HANDLER(STM32_ADC1_HANDLER) {

Then there is no need to make change to makefile AND...
 **_the time command now works again_** in the console command.

Why does removing space work but not commenting out line?

nlroth, 2021-04-22:
Whatever was changed after 1.0.46, fixed the time setting issue so I am closing this.

closed 2021-04-22 by nlroth