DiSlord/NanoVNA-D #8: Font is too small for F072 devices

salyzyn, 2021-06-12:
I have a workable solution for using an 8x8 font instead of 5x7 for the F072 NanoVNA-H devices. PTAL.
[Font8x8.zip](https://github.com/DiSlord/NanoVNA-D/files/6642302/Font8x8.zip)

DiSlord, 2021-06-15:
Possible build fw vs big font (need set for LCD_320x240) in nanovna.h
// Used marker size settings
#define _USE_BIG_MARKER_     1
// Used font settings
#define _USE_FONT_           1

But need some fixes for text position (and possible button size)

salyzyn, 2021-06-20:
The 8x8 fonts I copied in interfered a lot less, there were only a few places where I fixed the positions or content to make it OK the 7x13 fonts interfered a lot more and was a different beast comparatively. I am relatively happy where I've got it, including the ability on the NanoVNA-H device to be able to switch from 5x7 or 8x8 fonts on the fly in a Font Size menu item was useful.

FYI: While I was at it, playing in the code, I also added AM (25%, 50%, 75% and 100%) and FM (deviation from 0 to MHz) modulation for CW mode (made CW FREQ go into a submenu). The modulation was modulo-4 over the sweep points. The modulation tone worked out to roughly 500hz +/-  (at 4KHz BW setting) based on the sweep calculation and delay times, but good enough as a signal generator to tune my radios; although the inter-sweep drawing deadtime sucked a little.

I can clean things up (make 8x8 fonts like the 5x7 table rather than raw), how do you suggest I post the changes so you can review and/or integrate them?

Ho-Ro, 2021-07-05:
@salyzyn Nice contribution! Would you mind putting your development work into a public GitHub repo? That would make it much easier to track your progress.

DiSlord, 2021-11-30:
Added 6x10 font for small screens

closed 2021-11-30 by DiSlord