DiSlord/NanoVNA-D #10: Cannot build without SD card support

Ho-Ro, 2021-06-15:
I comment out the SD card support in nanovna.h (cloned from your current github):
````
// Add SD card support, req enable RTC (additional settings for file system see FatFS lib ffconf.h)
// #define __USE_SD_CARD__
// Allow run commands from SD card (config.ini in root)
// #define __SD_CARD_LOAD_
````
After this I get a lot build errors because of missing symbols.

**Reason:** `#ifdef __USE_SD_CARD__` in ui.c disables too many lines of code, e.g. `menu_back`.

**Fix:** move the needed code out of the `#ifdef ... #endif`
main.c:
````
@@ -95,11 +95,13 @@ static volatile vna_shellcmd_t  shell_function = 0;
 // Enable debug for console command
 //#define DEBUG_CONSOLE_SHOW
 // Enable usart command
 #define ENABLE_USART_COMMAND
 // Enable SD card console command
+#ifdef __USE_SD_CARD__
 #define ENABLE_SD_CARD_CMD
+#endif
````
ui.c:
````
@@ -1242,10 +1242,17 @@ static UI_FUNCTION_CALLBACK(menu_sdcard_cb)
   drawMessageBox("SAVE TRACE", res == FR_OK ? fs_filename : "  Fail write  ", 2000);
   request_to_redraw(REDRAW_AREA);
   ui_mode_normal();
 }
 
+static const menuitem_t menu_sdcard[] = {
+  { MT_CALLBACK, SAVE_S1P_FILE, "SAVE S1P", menu_sdcard_cb },
+  { MT_CALLBACK, SAVE_S2P_FILE, "SAVE S2P", menu_sdcard_cb },
+  { MT_NONE,     0, NULL, menu_back } // next-> menu_back
+};
+#endif
+
 #ifdef __DIGIT_SEPARATOR__
 static UI_FUNCTION_ADV_CALLBACK(menu_separator_acb)
 {
   (void)data;
   if (b){
@@ -1272,17 +1279,10 @@ static UI_FUNCTION_CALLBACK(menu_clean_trace_cb)
 static const menuitem_t menu_back[] = {
   { MT_CANCEL,   0, S_LARROW" BACK", NULL },
   { MT_NONE,     0, NULL, NULL } // sentinel
 };
 
-static const menuitem_t menu_sdcard[] = {
-  { MT_CALLBACK, SAVE_S1P_FILE, "SAVE S1P", menu_sdcard_cb },
-  { MT_CALLBACK, SAVE_S2P_FILE, "SAVE S2P", menu_sdcard_cb },
-  { MT_NONE,     0, NULL, menu_back } // next-> menu_back
-};
-#endif
-
 static const menuitem_t menu_calop[] = {
   { MT_ADV_CALLBACK, CAL_OPEN,  "OPEN",  menu_calop_acb },
   { MT_ADV_CALLBACK, CAL_SHORT, "SHORT", menu_calop_acb },
   { MT_ADV_CALLBACK, CAL_LOAD,  "LOAD",  menu_calop_acb },
   { MT_ADV_CALLBACK, CAL_ISOLN, "ISOLN", menu_calop_acb },

````

DiSlord, 2021-06-15:
Thanks, soon add this fix

Ho-Ro, 2021-06-30:
One additional change is needed if you build with SD card:
`static const menuitem_t menu_back[]` must be defined before usage, so it should be moved in front of the  
`#ifdef __USE_SD_CARD__` part.
````
// Back button submenu list
static const menuitem_t menu_back[] = {
  { MT_CANCEL,   0, S_LARROW" BACK", NULL },
  { MT_NONE,     0, NULL, NULL } // sentinel
};

#ifdef __USE_SD_CARD__
#define SAVE_S1P_FILE  1
#define SAVE_S2P_FILE  2
...
````

closed 2021-07-04 by DiSlord