DiSlord/NanoVNA-D #17: Build error when __VNA_Z_RENORMALIZATION__ is defined, include file missing

Ho-Ro, 2021-07-16:
Enabling `#define __VNA_Z_RENORMALIZATION__` gives this error during make:
````
main.c:1116:10: fatal error: vna_modules/vna_renorm.c: No such file or directory
 #include "vna_modules/vna_renorm.c"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
main.c:1116:10: fatal error: vna_modules/vna_renorm.c: No such file or directory
 #include "vna_modules/vna_renorm.c"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
````

DiSlord, 2021-07-17:
I not add this feature to release, and not want open source
This function leave only for internal use

DiSlord, 2021-07-17:
You can join to
https://groups.io/g/nanovna-beta-test
As tester for some features

closed 2021-07-17 by DiSlord