DiSlord/NanoVNA-D #18: Are the latest releases posted to this repository (eg v1.0.69) or another public repo?

owenduffy, 2021-08-07:


solhuebner, 2021-08-08:
A new release is usually posted when he is happy with the quality. There is a beta group for testing but releasing test versions to the public is not intended to avoid user frustration :)

owenduffy, 2021-08-08:
I've got it, The issues I have with v1.0.69 are not worth mention... it is a pre-release.

closed 2021-08-08 by owenduffy

solhuebner, 2021-08-08:
If you want to test, I wound recommend to join https://groups.io/g/nanovna-beta-test

owenduffy, 2021-08-08:
I am not into social media. I did test, but I have already uninstalled it based on it being pre-release.

Ho-Ro, 2021-08-08:
> The issues I have with v1.0.69 are not worth mention... it is a pre-release.

But that's exactly why pre-releases exist - test them and report problems to improve code quality.
Open source is not just taking - it also comes down to giving something back, either actively (code, documentation, translation) or passively (suggesting new features or reporting bugs).

Just my 2¢
Martin