DiSlord/NanoVNA-D #20: H4 v1.0.69 - Smith chart noisy after calibration

mooredan, 2021-09-12:
Branch: H4
version: 1.0.69
Latest commit:

commit 5d2ddda884409294d81f5e05f350c7e585458172 (HEAD -> H4, origin/H4)
Merge: 672e8e3 fe83afd
Author: DiSlord <dislord@mail.ru>
Date:   Tue Aug 31 21:15:17 2021 +0300

    Merge branch 'master' of https://github.com/DiSlord/NanoVNA-D into H4

Compiled and install firmware, did a "Clear all and reset". The display looks normal.
[image]

...and the version screen:
[image]

Did a Open-Short-Load calibration, upon hitting "DONE", the S11 LOGMAG and SMITH go "noisy" with the Smith Chart being a star-like pattern.

[image]

I want to think that I'm just missing something.

Shipped version 0.5.0 (hugen79) doesn't have this issue, reinstalled 0.5.0 after 1.0.69 to verify

DiSlord, 2021-09-13:
Check again calibration standarts (more better need check LOAD) and used cables

Not see any problems for disabled calibration.

mooredan, 2021-09-13:
I did install an earlier version and the problem went away (this was hugen79's 1.0.45 Jan 28).  Then I went back to this repo's 1.0.69 and the problem is gone.    Along the way I did change the calibration configuration and finally calibrated at the connectors of the NanoVNA to eliminate that possibility.  The "issue" is now resolved, but I cannot say if this was a software or hardware issue with any certainty.  Sorry for the distraction.

closed 2021-09-13 by mooredan