DiSlord/NanoVNA-D #23: Touchscreen calibration broken in latest master

W3AXL, 2021-12-03:
I have a NanoVNA-H, and I've built the latest firmware from the master branch and loaded it onto the device.

However, the touchscreen is now unable to calibrate the right quarter of the screen, despite repeated attempts to properly calibrate it via the menu. Below is a photo of the touch test screen after moving my stylus over as much of the screen as possible.

[image]

Interestingly, if I switch back to [edy555's last release from November 2020](https://github.com/ttrftech/NanoVNA/releases/tag/0.8.0), I am able to calibrate the screen properly and the full extent is usable. So, something strange is occurring with this branch of the firmware.

nlroth, 2021-12-03:
You've cracked the conductive strip along one edge of the touch surface. I've seen this many times in the past when PDAs used resistive touch surfaces.
You have 4 options: replace the entire LCD module, carefully remove and replace just the touch surface, use the unit only with the jog switch and finally, sell it and get a new unit.
I will say that if you take the housing off and carefully inspect the top or bottom edge of the touchscreen, you may see where the crack is and try using some type of conductive paint to bridge the spot but you need to be very careful as the conductive line for the right side of the touch panel runs along one of those edges.
good luck!

W3AXL, 2021-12-03:
Originally I figured the same thing - that it was a hardware defect in the touchscreen. And it still might be I suppose. But if that was the case - why does the old build from edy fix the issue?

The entire touch screen responds to input properly and there aren't any dead zones. It's just the calibration is way off and for whatever reason there's no way to bring it back into cal properly.

DiSlord, 2021-12-03:
Make sure that you are calibrating correctly, you must first press * at the upper left corner, and then * at the lower right. It is necessary to press clearly, in one movement.

In theory, the default calibration should also be suitable, test it with it (for this you need to reset the config, this can be done through CONFIG->EXPERT SETTINGS->CLEAR CONFIG->CLEAR AND RESET)

nlroth, 2021-12-03:
Try flashing with v1.1 again, then using the jog sw, from the Expert menu, do a clearconfig, then touch cal, save config and test the touch.

W3AXL, 2021-12-03:
Morning update: I think there may be a hardware fault after all. Depending on the pressure at which I touch the screen, I can sometimes get valid touch input on that far right quadrant. If I tap extremely lightly, I can get it to recognize touch properly all the way to the edge. However any harder of a press and it jumps over to the left.

Still curious that edy's old firmware is much better, but I think either way a new touch panel is in order. Does anyone happen to know a good source?

nlroth, 2021-12-03:
Refer to the nanoVNA forum messages - Hugen at some point 2 years ago gave a source but no idea if that is still viable.
There is a file in the hack & mod section that shows how to add a 3 inch display to the unit.

closed 2021-12-03 by W3AXL

DiSlord, 2021-12-03:
Try to also remove the top cover, sometimes it happens that it presses on the display and this leads to distortion of the touch of the touch