DiSlord/NanoVNA-D #24: Enhancement: SD access over USB connection for H4

errolt, 2021-12-29:
Please add SD card access over USB connection, as that would reduce wear on SD card socket from "insert SD card, take screen shot, remove SD card, insert SD card into PC".

I don't know if there is available rom space for this though.

nlroth, 2021-12-29:
It's already there - check out the console commands:sd_readsd_listsd_delete
sd_read dumps binary data back to the console, so you'll have to write a short bit of code to execute the sd_read command and save it into a file on the PC.
There is not enough room in the Nano to do any other file commands from the PC

errolt, 2021-12-29:
Hmm, that is a start, but I meant SD access as a second USB endpoint that enumerates as a disk drive. In other words, when you plug the usb cable in it enumerates as a serial port and a disk drive, at the same time.

nlroth, 2021-12-29:
Yes, I know what you're talking about but as I said (and DiSlord mentioned quite a while back), there is not enough room for that in the NanoVNA-H and H4 ROM.
LiteVNA is a different beast - lots of ROM space there, so it might happen for that device but it all depends on DiSlord's priorities.

closed 2021-12-30 by errolt