DiSlord/NanoVNA-D #27: NanoVNA Firmware version 1.1.00 - Incomplete menu image

oridolfi, 2022-03-24:
I have just updated my device and found that the Display -> Scale -> Scale/Div and Reference Position menus are not being displayed properly. The right/upper corner of the image shows previous menus images and as result, some buttons for this settings are missing.

Hope this feedback helps to improve your astonishing software.

Best regards
Oscar

OneOfEleven, 2022-03-27:
Surprised to see DiSlord making any posts at all with the psychotic 'P' doing it's thing. Hope you're safe.

DiSlord, 2022-03-27:
Not understand problem, but look like Scale/Ref button not draw? You use SD card? If yes remove it from device

oridolfi, 2022-03-27:
I menat to say that upper-Right buttons on the Scale/Div and Reference windows are not drawn.  Instead, part of the previous windows is shown in that place (upper-right).  please,see the both screens' attached pictures.
I'm using NanoVNA Firmware version 1.1.00 

[image]
[image]

Hope this clarifies.
Thanks.

oridolfi, 2022-03-28:
To complement my previous post.
This is the dfu file I downloaded: (https://github.com/DiSlord/NanoVNA-D/releases/download/v1.1.00/NanoVNA.H.v1.1.00.dfu)
No SD in use.

nlroth, 2022-03-28:
There is no problem. You're in a keyboard numerical entry mode so only the keyboard is shown until you enter a number and press return or just enter the back key with no entry. Either will cause the keyboard to clear the screen and you'll be back to where you were with fully functioning menus.

DiSlord, 2022-03-28:
Yes keyboard draw under menu, for this keyboard no buttons at this place

closed 2022-03-28 by DiSlord