DiSlord/NanoVNA-D #28: USB enumeration

owenduffy, 2022-04-15:
It appears that two devices with v1.1.0 firmware cannot both obtain a COM port on Win10, the first one attached does and the second one seems not the enumerate fully and produce a COM port. I note they both have the same VID,PID (to be expected) and SERIAL which may be the root cause of this problem.

Is it feasible to have each new installed firmware create a random serial number which it retains for future use. Perhaps if it recognised the factory serial no, it replaces it with a random value which it uses on later starts.

Owen

owenduffy, 2022-04-15:
... or perhaps there is a unique id available from the ST parts?

DiSlord, 2022-04-17:
Strange i can use H and H4 device on my Windows 7
[image]
COM10 NanoVNA H
[image]
COM29 NanoVNA H4
[image]

owenduffy, 2022-04-17:
Hi,
I repeated the test, and it is working fine. One of them comes with with serial 400, the next one has a blank serial and enumerates properly getting a com port. I tried this several times a few days ago and it consistently failed... perhaps a reboot has cleaned stuff up. Sorry for the false start.

closed 2022-04-17 by owenduffy