DiSlord/NanoVNA-D #30: Increase number of Save-Recall profiles from 6 to 12 (for example)

igor-m, 2022-06-01:
Would be great to increase the number to 12.
When doing antenna measurements you need say 10 for short waves and 2 for VHF/UHF (and you need them calibrated separately).

igor-m, 2022-06-01:
I've been building from the sources (H4 here with the sdcard). In v1.2 I built and flashed in my H4 yesterday I can see "Save Config" and "Save Calibration" but none "Load" counterparts.. Ok, with your special dfu there is the "Load" and with it I can save/load all 6 slots under a single file name from the sdcard. Good!
One issue I see - it could be the Calibration may somehow depend on the Config. Would not be better to save both Calibration and Config under one name into one file, like for example "Save/Load Profile" such the user always gets a complete H4 setup? 
That would be a safer solution, imho..

DiSlord, 2022-06-02:
Config contain only not depend from measure options.
Some of them stored in backup registers (not reset on power off, but need inable remember state in expert settings and save config)

  uint32_t _harmonic_freq_threshold;
  int32_t  _IF_freq;
  int16_t  _touch_cal[4];
  uint8_t  _vna_mode; // stored in backup - various display settings like flip/grid values/dot and so
  uint8_t  _brightness; // 
  uint16_t _dac_value;
  uint16_t _vbat_offset;
  uint16_t _bandwidth; // stored in backup
  uint16_t _lcd_palette[MAX_PALETTE];
  uint32_t _serial_speed;
  uint32_t _xtal_freq;
  float    _measure_r;
  uint8_t  _lever_mode; // stored in backup
  uint8_t  _digit_separator;
  uint8_t  _band_mode;

igor-m, 2022-06-02:
I've flashed in your special v1.2 .dfu for H4 and it works, it seems.
We may close this as the Load Calibration will do that.
Nice work!

igor-m, 2022-06-04:
I saved 6 slots with HF bands under the name "A", and 6 slots with UHF bands under name "B". Both files have 16kB on the sdcard. When loading the configuration A I still see slots with UHF. It seems like it does not load. What I am doing wrong?

DiSlord, 2022-06-04:
Then save to SD card you save in file. If file exist, old file delete, new data write.
Select different filanames.

igor-m, 2022-06-04:
I've learned the Save Calibration saves (into a file on the sdcard) only the currently active slot. In order to save all my 12 slots I have to save all individual slots under their specific names (such I get 12 files on the sdcard). 
In order to replace a specific slot (in the flash) I have to Load Calibration (that loads one of the 12 files I choose and shows it on the display), and then I have to Save it to the flash to a specific slot (Calibrate->Save).

closed 2022-06-04 by igor-m