DiSlord/NanoVNA-D #31: build fails

n3xus1, 2022-06-08:
build fails see error 

Thanks

Compiling flash.c
./NANOVNA_STM32_F072/flash.c: In function 'flash_wait_for_last_operation':
./NANOVNA_STM32_F072/flash.c:3:10: error: 'FLASH' undeclared (first use in this function)
   while (FLASH->SR == FLASH_SR_BSY) {
          ^~~~~
./NANOVNA_STM32_F072/flash.c:3:10: note: each undeclared identifier is reported only once for each function it appears in
./NANOVNA_STM32_F072/flash.c:3:23: error: 'FLASH_SR_BSY' undeclared (first use in this function)
   while (FLASH->SR == FLASH_SR_BSY) {
                       ^~~~~~~~~~~~
./NANOVNA_STM32_F072/flash.c: At top level:
./NANOVNA_STM32_F072/flash.c:9:31: error: unknown type name 'uint32_t'
 static void flash_erase_page0(uint32_t page_address)
                               ^~~~~~~~
./NANOVNA_STM32_F072/flash.c: In function 'flash_unlock':
./NANOVNA_STM32_F072/flash.c:22:3: error: 'FLASH' undeclared (first use in this function)
   FLASH->KEYR = 0x45670123;
   ^~~~~
./NANOVNA_STM32_F072/flash.c: At top level:
./NANOVNA_STM32_F072/flash.c:26:24: error: unknown type name 'uint32_t'
 void flash_erase_pages(uint32_t page_address, uint32_t size)
                        ^~~~~~~~
./NANOVNA_STM32_F072/flash.c:26:47: error: unknown type name 'uint32_t'
 void flash_erase_pages(uint32_t page_address, uint32_t size)
                                               ^~~~~~~~
./NANOVNA_STM32_F072/flash.c:36:37: error: unknown type name 'uint16_t'
 void flash_program_half_word_buffer(uint16_t* dst, uint16_t *data, uint16_t size)
                                     ^~~~~~~~
./NANOVNA_STM32_F072/flash.c:36:52: error: unknown type name 'uint16_t'
 void flash_program_half_word_buffer(uint16_t* dst, uint16_t *data, uint16_t size)
                                                    ^~~~~~~~
./NANOVNA_STM32_F072/flash.c:36:68: error: unknown type name 'uint16_t'
 void flash_program_half_word_buffer(uint16_t* dst, uint16_t *data, uint16_t size)
                                                                    ^~~~~~~~
make: *** [build/obj/flash.o] Error 1

DiSlord, 2022-06-09:
Made 'make clean'
Also need remove flash.c from root (if file exist)

n3xus1, 2022-06-09:
the issue is still present

I tried with your instructions and also a fresh git clone and both give me the same error

thanks,

Compiling flash.c
./NANOVNA_STM32_F072/flash.c: In function 'flash_wait_for_last_operation':
./NANOVNA_STM32_F072/flash.c:3:10: error: 'FLASH' undeclared (first use in this function)
   while (FLASH->SR == FLASH_SR_BSY) {
          ^~~~~
./NANOVNA_STM32_F072/flash.c:3:10: note: each undeclared identifier is reported only once for each function it appears in
./NANOVNA_STM32_F072/flash.c:3:23: error: 'FLASH_SR_BSY' undeclared (first use in this function)
   while (FLASH->SR == FLASH_SR_BSY) {
                       ^~~~~~~~~~~~
./NANOVNA_STM32_F072/flash.c: At top level:
./NANOVNA_STM32_F072/flash.c:9:31: error: unknown type name 'uint32_t'
 static void flash_erase_page0(uint32_t page_address)
                               ^~~~~~~~
./NANOVNA_STM32_F072/flash.c: In function 'flash_unlock':
./NANOVNA_STM32_F072/flash.c:22:3: error: 'FLASH' undeclared (first use in this function)
   FLASH->KEYR = 0x45670123;
   ^~~~~
./NANOVNA_STM32_F072/flash.c: At top level:
./NANOVNA_STM32_F072/flash.c:26:24: error: unknown type name 'uint32_t'
 void flash_erase_pages(uint32_t page_address, uint32_t size)
                        ^~~~~~~~
./NANOVNA_STM32_F072/flash.c:26:47: error: unknown type name 'uint32_t'
 void flash_erase_pages(uint32_t page_address, uint32_t size)
                                               ^~~~~~~~
./NANOVNA_STM32_F072/flash.c:36:37: error: unknown type name 'uint16_t'
 void flash_program_half_word_buffer(uint16_t* dst, uint16_t *data, uint16_t size)
                                     ^~~~~~~~
./NANOVNA_STM32_F072/flash.c:36:52: error: unknown type name 'uint16_t'
 void flash_program_half_word_buffer(uint16_t* dst, uint16_t *data, uint16_t size)
                                                    ^~~~~~~~
./NANOVNA_STM32_F072/flash.c:36:68: error: unknown type name 'uint16_t'
 void flash_program_half_word_buffer(uint16_t* dst, uint16_t *data, uint16_t size)
                                                                    ^~~~~~~~
make: *** [build/obj/flash.o] Error 1

NanoVNA-D % ls
ChibiOS			build			ili9341.c		si5351.h
FatFs			chconf.h		lc_matching.c		spi.h
Font10x14.c		chprintf.c		main.c			tlv320aic3204.c
Font5x7.c		data_storage.c		mcuconf.h		ui.c
Font6x10.c		doc			nanovna.h		usbcfg.c
Font7x11b.c		dsp.c			numfont20x22.c		usbcfg.h
Makefile		dsp.h			plot.c			vna_math.c
NANOVNA_STM32_F072	halconf.h		prog.sh			vna_math.h
NANOVNA_STM32_F303	hardware.c		python
README.md		hardware.h		si5351.c

DiSlord, 2022-06-09:
I can forgot push last changes, try again

n3xus1, 2022-06-10:
Thanks,
 
It builds fine now, I load it a little later when I have some time.

:)

closed 2022-06-10 by DiSlord