DiSlord/NanoVNA-D #32: Fail to flash .dfu file

as3ii, 2022-06-28:
I've tried downloading the .dfu file from the release section, but when flashing with `dfu-util -d 0483:df11 -a 0 -s 0x08000000:leave -D NanoVNA.H4.v1.2.00.dfu` the program exit with an error.
```
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Multiple alternate interfaces for DfuSe file
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
dfu-util: Error: File ID 0483:0000 does not match device (0483:df11 or 0483:df11)
```

building the firmware from source and flashing the compiled file works (but the buttons to load the calibration file and config file are missing)

n3xus1, 2022-07-08:
this is the command I used to upload the .dfu file and worked for me. I'd try just renaming the file for your H4 and try again

`dfu-util -a 0 -D NanoVNA.H.v1.2.00.dfu`

as3ii, 2022-07-09:
this worked, thanks

closed 2022-07-09 by as3ii