DiSlord/NanoVNA-D #40: Please include an Intel hex format file in the release

owenduffy, 2022-09-21:
Lots of users have problems updating firmware, and one of the contributions is determination to use STM32CUBEPROGRAMMER which will not open .dfu files, but will open a .hex file. Part of the reason for this is that ST seem to have deprecated DFUSEDEMO, but have chosen to not support .dfu in STM32CUBEPROGRAMMER.

If the release included a .hex format, it would make the process simpler for those users.

(Yes, I am aware that STM32CUBEPROGRAMMER uses libusbk or like driver, and that is easily installed with Zadig.)

Keep up the good work.

Owen

Ho-Ro, 2022-09-22:
These users have a bunch of other choices:

- If you stick with downloaded `*.dfu` files, you can
  - use [`dfu-util`](http://dfu-util.sourceforge.net/) (binaries for Win are available) to transfer the file to the NanoVNA.
  - you can post-process the `*.dfu` with the Python script [`dfuse-extract.py`](https://github.com/craftit/dfuse-extract), which can _'Extract DfuSe images (.dfu) into plain binary files'_ that can easily be recoded as intel-hex with the tool of your choice - typically named `bin2hex`.
- If you build the FW yourself (this would be my recommended option, because you can fine-tune the behaviour of the device - just type `make`), you will get both `build/ch.hex` (which is what you want) and `build/ch.bin`, the latter format can be uploaded with `make flash` (on Linux).

Ho-Ro, 2022-10-19:
## UPDATE
@owenduffy et al.
I provide unmodified bin and hex binaries of DiSlord's latest version in a branch of my clone, for version info just check the commit message - currently version 1.2.15:
https://github.com/Ho-Ro/NanoVNA-D/tree/current-bin-hex

These are the download links:
- https://github.com/Ho-Ro/NanoVNA-D/raw/current-bin-hex/NanoVNA-H.bin
- https://github.com/Ho-Ro/NanoVNA-D/raw/current-bin-hex/NanoVNA-H.hex
- https://github.com/Ho-Ro/NanoVNA-D/raw/current-bin-hex/NanoVNA-H4.bin
- https://github.com/Ho-Ro/NanoVNA-D/raw/current-bin-hex/NanoVNA-H4.hex
---
@DiSlord: It would be helpful for many users if you could provide these files as an additional service, that's all you need:

```shell
make clean; TARGET=F072 make -j \
&& cp build/ch.bin NanoVNA-H.bin \
&& cp build/ch.hex NanoVNA-H.hex

make clean; TARGET=F303 make -j \
&& cp build/ch.bin NanoVNA-H4.bin \
&& cp build/ch.hex NanoVNA-H4.hex
```

owenduffy, 2024-04-12:
Thank you Martin, your frequent builds are a very useful resource... keep it up.

Owen

closed 2024-06-18 by DiSlord