DiSlord/NanoVNA-D #41: Artefacts when capturing the screen

Ho-Ro, 2022-11-13:
The screen capture shows artefacts at the bottom of the image in the far right column. This is independent of the respective device and the recording program used; tinySA also shows the same artefacts. NanoVNA version 1.2.15, the issue existed also with versions 1.0.64 and 1.1.x, also latest tinySA version. 

[image] [image]

[image] [image]

Another user's tinySA also shows this, so it is obviously a systematic problem.
https://groups.io/g/tinysa/message/7071

[image] [image]

DiSlord, 2022-11-14:
Here is NanoVNA V2 screenshot example (artefact also present)
[image]
Different hardware (CPU and it have different SPI module, different firmware)

If use st7789 (LiteVNA62) firmware (LCD part) fully equal no artefact
[image]
As can see no artefacts, so i think this LCD ili9341 driver hardware problem.

Artefact visible after made screenshot (read operation from LCD).

Ho-Ro, 2022-11-14:
Thx for your update, please close unless you want to investigate further.

closed 2024-08-21 by DiSlord