DiSlord/NanoVNA-D #43: S11 impedance changed with V1.2?

errolt, 2022-12-02:
With V1.1, a 50 ohm load showed +-50ohm and impedance in the S11 SMITH line:
[image]

But with V1.2 it seems to show some dimensionless value and angle?
[image]

This affects all my saved screen setups.

How do I switch back to restive + reactive display?

Ho-Ro, 2022-12-02:
Go to menu `DISPLAY` / `TRACE` (select the smith trace) / `<-BACK` / `FORMAT S11 (REFL)` / `SMITH / R + L/C`

errolt, 2022-12-02:
Wow. Thank you. I would never have figured out that flow.

closed 2022-12-02 by errolt