DiSlord/NanoVNA-D #49: make flash fails

n3xus1, 2023-01-10:
I receive this error when issuing make flash
 
% make flash
make: *** No rule to make target `build/build/H.bin', needed by `flash'.  Stop.

thanks,

Ho-Ro, 2023-01-10:
See my comment(s):
https://github.com/DiSlord/NanoVNA-D/commit/c60f20a82a32aed8e10a2683fc4d1fe662682e94#r95702779

n3xus1, 2023-01-10:
> See my comment(s):
> [c60f20a#r95702779](https://github.com/DiSlord/NanoVNA-D/commit/c60f20a82a32aed8e10a2683fc4d1fe662682e94#r95702779)
> [c60f20a#r95702904](https://github.com/DiSlord/NanoVNA-D/commit/c60f20a82a32aed8e10a2683fc4d1fe662682e94#r95702904)

I see your comments, so these changes need to be pushed still I take it.
thanks

Ho-Ro, 2023-01-11:
> I see your comments, so these changes need to be pushed still I take it.

@n3xus1 you could create a local branch and fix it on your own until fixed by upstream.

PR #51 (2023-01-13): fix #49 - make flash fails
fix #49 - make flash fails
commit 3149f0de (2023-01-13): Merge pull request #51 from Ho-Ro/fix_49

fix #49 - make flash fails
closed 2023-01-13 by DiSlord