DiSlord/NanoVNA-D #57: Oopsie.. LiteVNA64 flashing

hvegh, 2023-03-26:
So I figured my luck with the updated firmware using <master>  branch and TARGET=F303:

Put the device in DFU mode by pulling BOOT0 HIGH and flashed the H4 image, see update procedure below.
Screen stays dark.. 

Any ideas maybe how to fix? 

```
~ # dfu-util -l
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [2e3c:df11] ver=0200, devnum=32, cfg=1, intf=0, path="2-2", alt=1, name="@Option Byte   /0x1FFFF800/01*048 e", serial="AT32"
Found DFU: [2e3c:df11] ver=0200, devnum=32, cfg=1, intf=0, path="2-2", alt=0, name="@Internal Flash  /0x08000000/ 512*2Kg", serial="AT32"

```

So i went and went ahead and flashed :

```
~# dfu-util -d 0483:df11 -a 0 -s 0x08000000:leave -D build/H4.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 2e3c:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(10) = dfuERROR, status(14) = Something went wrong, but the device does not know what it was
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 85704
Erase   	[=========================] 100%        85704 bytes
Erase    done.
Download	[=========================] 100%        85704 bytes
Download done.
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state
```

hvegh, 2023-03-27:
Additional hardware info:

LiteVNA64 hw. ver. 0.3

Chips used:

```
AT32F403 - Microcontroller
BN24.0   - 24MHz xtal
MS5351M  - clock generator
MAX2871E - Freq. generator
HR2046   - XPT2046 Touch Screen Controller
GS8722   - dual 11MHz Opamp
8641     - RF switch 3.4.3.4 pins
LN4056-H - Linear technology lion charger
```

DiSlord, 2023-03-27:
This source and firmware only for H or H4 hardware.
LiteVNA use different code (but at this moment most hardware independent part equal, i port this code here) and not open source.

You need install LiteVNA64 firmware 
https://www.zeenko.tech/litevna
Or use my build
https://groups.io/g/nanovna-beta-test/message/4224

DiSlord, 2023-03-27:
I open this thread. As i can see you erase LiteVNA bootloader (not shure, as i know AT32 CPU not allow flash from STM32  DFU util)
Suggest go to this thread
https://groups.io/g/nanovna-beta-test/message/4224
Try help restore your device

closed 2023-03-27 by DiSlord

hvegh, 2023-03-27:
Will check it, thank you!

DiSlord, 2023-03-27:
Here is instuction how restore LiteVNA after full brick (restore bootloader and firmware)
https://groups.io/g/nanovna-beta-test/message/3982

hvegh, 2023-03-27:
Ah yes! 

I have restored an initial firmware backup using:
dfu-util -d 0483:df11 -a 0 -s **0x08004000**:leave  -D ../archive/VNA_v1.2.bin 

The AT32 microconroller needs extra 0x4000 offset...

Solved!