DiSlord/NanoVNA-D #60: Bug in NanoVNA firmware - XTAL calculations

Ho-Ro, 2023-05-05:
Hello DiSlord,
there was an [interesting discussion](https://groups.io/g/nanovna-users/topic/bug_in_nanovna_firmware/98691046) about the xtal calculation on groups.io. Please check Orestes' findings.
Martin

orestesmas, 2023-05-07:
TL;DR

Essentially, NanoVNA does XTAL measures (and lc-series too, by the way) following the formulae provided in [this article](https://www.mikrocontroller.net/attachment/473317/Crystal_Motional_Parameters.pdf), but these formulae are not applicable as-is in 

Suppose you use the following measurement circuit:

[image]

Then the voltages measured by NanoVNA are Vin and Vo, so the leftmost Ro has to be left out of the calculations. The total effective resistance is, thus, Ro + Rm, The amplification at the resonant frequency is given by the voltage divider Ro/(Rm+Ro) and the filter's (loaded) bandwidth is (Rm+Ro)/Lm. The resonant frequency in Hz is 1/(2·pi·sqrt(Lm·Cm)).

From these relations you can derive the values of Rm, Lm and Cm. Then you can use the minimum of amplification at the parallel resonance to determine Cp

I can provide a patch through a pull request if necessary.

Ho-Ro, 2023-05-12:
Orestes, I'm no more sure if we can ignore Ro as in your schematics above.
When I look at the NanoVNA schematics the generator is measured directly via the yellow path while port 1 (your vin(t) signal) is fed via the bridge in the green circle.
[image]

orestesmas, 2023-05-13:
First: sorry for delaying this. I'm in the middle of final exams (as a teacher) and my spare time is reduced now.

Yes. I'm aware of the schematics. It seems that NanoVNA is able to measure the generator directly (the "reference" path) and also the voltage on Port 1 ("Reflect" path) and Port 2 ("Transmission" path).

So, I've to study the firmware's code to see how the measurement of S21 is done exactly. Depending on the answer, then maybe I'm right or wrong. (*)

So please let's put this issue on standby until we have more certainties.

Regards,
Orestes.

--------------------
(*) The ultimate question is how is S21 defined, because this quantity is related to the "insertion loss". 

When I was reading information about NanoVNA (I'm a month-old user of this device) I've read on the internet many sites that define (perhaps erroneously) S21 as the "voltage gain" between the input and the output. According to this, S21 is defined as the ratio between V1 (port 1 voltage) and V2 (port 2 voltage) MEASURED SIMULTANEOUSLY.  I assumed (perhaps erroneously) that NanoVNA does that and, if so, then I'm right that the calculations NanoVNA does are wrong because the source impedance must be out of the calculations.

On the other side, regarding to Wikipedia, one can define

"INSERTION LOSS:
In telecommunications, insertion loss is the loss of signal power resulting from the insertion of a device in a transmission line or optical fiber and is usually expressed in decibels (dB). 

If the power transmitted to the load before insertion is PT and the power received by the load after insertion is PR, then the insertion loss in decibels is given by,

    IL (dB) = 10 log10 PT/PR = 20 log10 |V1|/|V2|

Where, in this case, |V1| is the signal level WITHOUT the D.U.T. installed and |V2| is the signal level WITH the filter installed. Note that in this case, V1 and V2 are signals measured AT DIFFERENT TIMES.

If the two measurement ports use the same reference impedance (which is the case in NanoVNA), then the IL and S21 are related:

IL = -20 log10 |S21|

If that's true then NanoVNA calculations are right and I'm wrong, because the power gain is defined which respect to the power in the maximum power transfer condition. In this case, the source impedance has to be taken into account.

To decide which is the case, I've to look into the firmware's code to determine how NanoVNA calculates the S21 parameter.

DiSlord, 2023-05-14:
>  V1 and V2 are signals measured AT DIFFERENT TIMES
NanoVNA measure Reference voltage and Thru voltage in one time (use 2 port audio codec)
V2/Lite series measure in different time (use one ADC port), but signals phase aligned (measure period = N * IF period), by this reason XTAL measures not fully correct, but ~ok if use big AVG settings)