DiSlord/NanoVNA-D #63: Cable length measurement v1.2.22

AxxiMator, 2023-05-17:
When measuring for a cable, it shows for a cable 10 times more than the real one. Velocity is set to 66%, with a real cable length of about 7 meters, the NanoVNA reading is about 70 meters.

AxxiMator, 2023-05-17:
Так  на каком частотном промежутке измерят длину кабеля? 50-900 - одна длина, 50-800 другая. 
Непонятна фраза по мере увеличения длины уменьшайте верхнюю частоту - зачем и как определить, при каких значениях верхней частоты прибор показывает корректную длину кабеля?

nlroth, 2023-05-17:
Refer to the user group.You can start with this msg thread: https://groups.io/g/nanovna-users/topic/77375866#26884
Do a search for TDR on the user group for more information.

AxxiMator, 2023-05-18:
Thank you very much. I will study further.

closed 2023-05-18 by AxxiMator