DiSlord/NanoVNA-D #65: Buggy firmware when compiling with GCC-ARM version 12 - ok with GCC-ARM version 8

Ho-Ro, 2023-07-01:
Compiling the tinySA source code with debian stable `gcc version 12.2.1 20221205 (15:12.2.rel1-1)` gives a [buggy firmware](https://github.com/erikkaashoek/tinySA/issues/70) - latest tinySA version as well as older versions.

**Compiling the NanoVNA-H firmware gives comparable warnings caused by ChibiOS.**
(I did not download this version to my NanoVNA-H because I assume that it also leads to a faulty device).

After downgrade to the debian oldstable version `gcc version 8.3.1 20190703 (release) [gcc-8-branch revision 273027] (15:8-2019-q3-1+b1)` it compiles fine.

**Warnings when compiling with GCC-ARM version 12:**
```
Compiling crt1.c
Compiling vectors.c
Compiling chsys.c
Compiling chdebug.c
Compiling chtrace.c
Compiling chvt.c
Compiling chschd.c
Compiling chthreads.c
Compiling chcore.c
Compiling chcore_v6m.c
In file included from ChibiOS/os/common/ext/CMSIS/ST/STM32F0xx/stm32f072xb.h:129,
                 from ChibiOS/os/common/ext/CMSIS/ST/STM32F0xx/stm32f0xx.h:157,
                 from ChibiOS/os/common/startup/ARMCMx/devices/STM32F0xx/cmparams.h:72,
                 from ChibiOS/os/common/ports/ARMCMx/chcore.h:70,
                 from ChibiOS/os/rt/include/ch.h:81,
                 from ChibiOS/os/common/ports/ARMCMx/chcore_v6m.c:28:
In function '__set_PSP',
    inlined from 'NMI_Handler' at ChibiOS/os/common/ports/ARMCMx/chcore_v6m.c:72:3:
ChibiOS/os/common/ext/CMSIS/include/core_cm0.h:85:28: warning: listing the stack pointer register 'sp' in a clobber list is deprecated [-Wdeprecated]
   85 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler          */
      |                            ^~~~~
ChibiOS/os/common/ext/CMSIS/include/core_cmFunc.h:443:3: note: in expansion of macro '__ASM'
  443 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
ChibiOS/os/common/ext/CMSIS/include/core_cm0.h:85:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   85 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler          */
      |                            ^~~~~
ChibiOS/os/common/ext/CMSIS/include/core_cmFunc.h:443:3: note: in expansion of macro '__ASM'
  443 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
In function '__set_PSP',
    inlined from '_port_irq_epilogue' at ChibiOS/os/common/ports/ARMCMx/chcore_v6m.c:125:5:
ChibiOS/os/common/ext/CMSIS/include/core_cm0.h:85:28: warning: listing the stack pointer register 'sp' in a clobber list is deprecated [-Wdeprecated]
   85 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler          */
      |                            ^~~~~
ChibiOS/os/common/ext/CMSIS/include/core_cmFunc.h:443:3: note: in expansion of macro '__ASM'
  443 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
ChibiOS/os/common/ext/CMSIS/include/core_cm0.h:85:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   85 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler          */
      |                            ^~~~~
ChibiOS/os/common/ext/CMSIS/include/core_cmFunc.h:443:3: note: in expansion of macro '__ASM'
  443 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
Compiling osal.c
Compiling hal.c
Compiling hal_st.c
Compiling hal_buffers.c
Compiling hal_queues.c
Compiling hal_mmcsd.c
Compiling hal_ext.c
Compiling hal_gpt.c
Compiling hal_i2c.c
Compiling hal_i2s.c
Compiling hal_pal.c
Compiling hal_serial.c
Compiling hal_serial_usb.c
Compiling hal_spi.c
Compiling hal_usb.c
ChibiOS/os/hal/src/hal_usb.c: In function '_usb_ep0in':
ChibiOS/os/hal/src/hal_usb.c:842:8: warning: this statement may fall through [-Wimplicit-fallthrough=]
[... 119 lines cut ...]
      |                          ~~~~~~~^~~~~
Compiling plot.c
plot.c:903:1: warning: 'cell_drawstring' defined but not used [-Wunused-function]
  903 | cell_drawstring(char *str, int x, int y)
      | ^~~~~~~~~~~~~~~
Compiling ui.c
Compiling ili9341.c
Compiling numfont20x22.c
Compiling Font5x7.c
Compiling Font7x13b.c
Compiling Font10x14.c
Compiling flash.c
Compiling adc.c
Compiling rtc.c
Linking build/ch.elf
Creating build/ch.hex
Creating build/ch.bin
Creating build/ch.dmp

   text    data     bss     dec     hex filename
  79816     528   16048   96392   17888 build/ch.elf
Creating build/ch.list

Done
```

DiSlord, 2023-07-01:
Most of warnings not critical and related to ChibiOS

I use gcc v9.3.1 for compile, this version provide more compact code:
   text	   data	    bss	    dec	    hex	filename
  86668	    648	  40728	 128044	  1f42c	build/H4.elf

If use v12.2.1 result:
   text	   data	    bss	    dec	    hex	filename
  87216	    648	  40728	 128592	  1f650	build/H4.elf

PS I use Eclipse xPack, this allow easy select compiler version
PSS upload firmware builded by v12.2.1 on my H4 device, and it works well.

Remove warnings in tis commit
https://github.com/DiSlord/NanoVNA-D/commit/ff96b9cc595f43e61d0d73acd0f1844fe3947024

Ho-Ro, 2023-07-01:
Thx, I will try, but until [tinySA is fixed](https://github.com/erikkaashoek/tinySA/issues/70#issuecomment-1616059765) I'll stay at the debian oldstable version 8.

Ho-Ro, 2023-07-02:
I installed v8..v12 in parallel and can select one of these versions for my builds. Starting with version 9 these "ChibiOS" warnings appear, would an update of ChibiOS be required for the newer compiler versions?

Ho-Ro, 2024-07-09:
To fix the ChibiOS warnings change [this line](https://github.com/DiSlord/NanoVNA-D/blob/94f03ec542604e512fbfa9ffff41e90a63160d4f/ChibiOS/os/common/ext/CMSIS/include/core_cmFunc.h#L443)
```c
  __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
```
to
```c
  __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) :);
```
and [this line](https://github.com/DiSlord/NanoVNA-D/blob/94f03ec542604e512fbfa9ffff41e90a63160d4f/ChibiOS/os/common/ext/CMSIS/include/core_cmFunc.h#L470)
```c
  __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
```
to
```c
  __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) :);
```

**Reason**
Starting from gcc v9 it is deprecated that the SP is part of the clobber list, even if the SP is changed after the ASM code is called (because that is the purpose of the command *"set SP"*). Newer ChibiOS versions just omit the clobber list in both commands.

closed 2024-08-17 by DiSlord