DiSlord/NanoVNA-D #68: Unable to read .cal files on v1.2.20: "Format err"

beaugunderson, 2023-09-22:
Hi there; I upgraded to v1.2.20 so I could save calibrations to the SD card and have more slots.

I initially tried with exFAT, I can save the files but my NanoVNA-H4 (ZeenKo brand) will not read them. I then tried formatting as FAT and saw identical behavior (I can write the files but not read them back). It always fails with "Format err".

Looking at the files in a hex editor the magic looks wrong, though I'm not sure if there's encoding that happens so maybe I'm just missing it.

I've attached a FAT version and an exFAT version of a calibration file I saved that will not load.

[Calibrations.zip](https://github.com/DiSlord/NanoVNA-D/files/12694437/Calibrations.zip)

beaugunderson, 2023-09-24:
I have now acquired a 32gb card, formatted it with FAT using the "SD Card Formatter" app, and I still get "Format err" upon reading.

beaugunderson, 2023-09-24:
Managed to add some additional error messages to the firmware, build it, and I see that the magic in the loaded file is 0, which corresponds to what's actually in the file:

[image]

So this does not seem like an SD issue but moreso an issue with magic being unset in the properties?

I will keep debugging with my limited C skills...

beaugunderson, 2023-09-24:
OK, so this does appear to be an issue with setting `magic` in the properties that are written to the file. If I use a hex editor and replace the 4 start bytes with `52 4e 4f 43` my device can then read the file without issue.

beaugunderson, 2023-09-24:
This was helpful in narrowing down the issue... my C is not good enough to figure out how to do this without the need for a second error variable that I can `sprintf` to, but you get the picture:

[image]

beaugunderson, 2023-09-24:
Disabling the magic check also got me going at least, in terms of being able to use the functionality. :)

DiSlord, 2023-09-24:
I think better fill magick field here
```
// Load propeties default settings
static void load_default_properties(void) {
//Magic add on caldata_save
  **current_props.magic             = CONFIG_MAGIC;**
```
This allow fill field for saved file, and not give error on load
https://github.com/DiSlord/NanoVNA-D/commit/be2fe7cff5d18eae5c3cb2777993192652f81191

beaugunderson, 2023-09-24:
> be2fe7c

it should be PROPERTIES_MAGIC though right? that’s what the loading code
checks for

DiSlord, 2023-09-24:
Yes i fix this typo in next commit (old commented code contain wrong magic)

beaugunderson, 2023-09-24:
awesome, thanks!

closed 2023-10-10 by DiSlord