DiSlord/NanoVNA-D #74: Feature request: Screen OFF serial command

sbridger, 2023-12-29:
(Firstly, I have a V2, so perhaps I can't use this anyway, so it's more a suggestion than a request)

When running remote serial, (I assume that) the display is a large part of the supply current. It would be good to turn it off. 
It would also be good to be able to deep sleep the unit and wakeup on pin change on the serial in.

Why: When testing antennas, it is desirable to have no cables, and no transmitter (bluetooth module). This can be easily acheived with TOSLINK optic fibre serial connection. However this means it operates off battery power.

DiSlord, 2024-01-03:
Made some test on my nanoVNA H4 power usage:
Set Power 2ma (for constant generator power usage)
Now set backlight value (config->brighntess):
brighntess 100% 0.367mA
brighntess 50% 0.312mA
brighntess  0% 0.253mA

Next I set backlight to 0 (but hardware on allow off backlight)
Additional I send LCD module off command and disable any display update
0.231 mA

As can see most good just set brighntess to 0%

DiSlord, 2024-01-03:
Most power usage on V2 - ADF4350 or MAX2870 generators, LCD display module need ~20mA power + power for backlight (you can set to 0 in config for minimise power usage)

If you set sweep range to SI5351 range ( < 140MHz) then V2 disable ADF4350 / MAX2870 generators

sbridger, 2024-01-03:
Thanks for your reply, 
Please excuse this question, but use of your firmware for V2 is recommended, but it seems you do not maintain  [V2 version](https://github.com/DiSlord/NanoVNA-V2-firmware)?
Does this NVNA-D F/W work with V2?

DiSlord, 2024-01-03:
My variantV2 firmware not open
Better say my variant for LiteVNA firmware not open, privided by me V2 firmware based on LiteVNA code, nanoVNA-D for H / H4 also use parts of LiteVNA code (and open)

closed 2024-02-04 by DiSlord