DiSlord/NanoVNA-D #81: Raw s21 phase at low frequencies

owenduffy, 2024-06-08:
I am using NanoVVNA-H4 v4.3 and firmware v1.2.30.
I expected that raw (ie uncorrected) s21 phase would be close to 0°.

[image]

Instead, it is close to 180°.

Looking at the schematic:

[image]

The polarity of Port 2 connection to the mixer is opposite to Port 1 which I assume needs to be undone in the firmware... but is it?

Owen

DiSlord, 2024-06-08:
Yes, measure S21 L/R pins swapped on AIC codec input, not possible change this by software
Need swap RAW data, but this need more processor time
Anyway this fixed by calibration

owenduffy, 2024-06-08:
>...
> Looking at the schematic:
> ...
> The polarity of Port 2 connection to the mixer is opposite to Port 1 which I assume needs to be undone in the firmware... but is it?

Rechecking the schematic, the polarity of the signal path for mixers U6 and U8 appear correct, yet measurement of raw s21 saved by the firmware is of opposite  phase to that expected as in the screenshot.

Nevertheless, raw s21 is opposite phase to expected as show in the screenshot.

Yes, correction masks the problem... but if one wants to extract raw measurements to correct elsewhere, then all the s21 figures have to been inverted to make sense.

So the question is why is s21 inverted, and it seems as I understand it, the fix involves multiplying the raw Port 2 measurement complex value by -1, so real and imaginary parts each *-1... or undoing where they are incorrectly inverted in the current firmware?

The discussion at [An experiment with NanoVNA and series through impedance measurement](https://owenduffy.net/blog/?p=32634) shows that the 'corrected s21' from the firmware is flawed which leads to the reported values of series through R and X being wrong.

Owen

DiSlord, 2024-06-09:
I double-checked everything and didn't find any errors.
Measurements of S11 and S21 take place along the same path, the inputs are switched in the audio codec. Reference channel IN2 is always connected to the left channel of the codec, and IN1 (reflected signal) to the right, alternately with IN3 (THRU)

At the same time, when changing channel 1, there are no problems with the phase, but in channel 3 it is rotated 180 degrees, as if the differential inputs R and L were reversed. There is no way to programmatically connect them differently

owenduffy, 2024-06-09:
Thanks for the checking.

Let me try to explain the schematic given earlier.

Let's look at the AC voltage applied to mixer U6 (call it Vs), pin A is high, pin B is low so Va-Vb is approximately in phase with the CLK1 voltage.

Let's put a short on the TX jack, so the AC voltage on mixer U7 (call it Vr) pin B is low and pin A is high... so Va-Vb is approximately in phase with the CLK1 voltage.

Now this Vr is in phase with Vs, so to correctly calculate s11 (which is -1 for a short), the code must calculate -Vr/Vs. (The raw s11 saved to a .s1p file for short is correctly a -ve value.) Somewhere s11 calculation must include an inversion, as you note it is not in the schematic.

Now look at mixer U8 used for s21. Va-Vb is approximately in phase with RX voltage, and in phase with TX voltage and CLK1 if Port 1 is connected to Port 2 with a very short cable... so no inversion is required to calculate s21.

Perhaps what is happening is that both s11 and s21 calcs invert when only the s11 should invert (ie multiply by -1) and s21 should be non-inverting?

owenduffy, 2024-06-13:
It appears that this is fixed in v1.2.32???

Owen

franciscosalvat, 2024-07-01:
where do I download version 1.32? I only see 1.2.29, thanks

Ho-Ro, 2024-07-09:
To get the latest version it is a good idea to [build it on your own](https://github.com/DiSlord/NanoVNA-D#build-the-nanovna-h-firmware). Just follow the instructions of the `README.md`. The [current GitHub version](https://github.com/DiSlord/NanoVNA-D/blob/94f03ec542604e512fbfa9ffff41e90a63160d4f/main.c#L128), i.e. at the time of writing this comment, is
```c
#define VERSION "1.2.36"
```