DiSlord/NanoVNA-D #82: Series through measurement of a known 200 ohm resistor yields poor result

owenduffy, 2024-06-11:
I conducted an experiment to measure a 200 ohm 1% resistor in a fixture with very short connections.

At 1MHz it measured R=225 and X~=0 by series through s21, but very close to 200+j0 by s11 reflection.

I then saved raw SOLT files and a raw s21 sweep and used scikit-rf to develop a one path calibration and apply it to the raw sweep. The results were much closer to expectation.

It appears that NanoVNA-H4 v4.3 and firmware v1.2.30 have a problem with making corrected s21 measurements.

Whilst it seems that s21 measurements of matched unknows like a 10dB attenuator gives expected result, the mismatched nature of the series 200 ohm DUT is not properly corrected. Also noted is that a s11 reflection measurement of the same result gave expected results suggesting the correction for source mismatch is working properly in that application.

The experiment is described in detail at [An experiment with NanoVNA and series through impedance measurement](https://owenduffy.net/blog/?p=32634) and [An experiment with NanoVNA and series through impedance measurement… more](https://owenduffy.net/blog/?p=32699).

Inserting a good 10dB attenuator on the Port 2 jack and calibrating made no significant difference, but adding the same attenuator to Port 1 and recalibrating did improve accuracy, again pointing at failure of the source match correction in the firmware.

Owen

DiSlord, 2024-06-12:
Due to NanoVNA not full 2 port VNA, it  use Incomplete_2-Port_Vector_Network_Analyzer_Calibration
https://www.researchgate.net/publication/263307473_Incomplete_2-Port_Vector_Network_Analyzer_Calibration_Methods

For S21 NanoVNA use 1-PORT + NORMALIZATION method, this method give errors
ENHANCED RESPONSE method gieve better results, but need more space for calibration data and more calculations

Main error from not fully 50Om Ch1 (uncnown params)

PR try use NanoVNA Saver calibration (reset calibration on NanoVNA) and measure S21 (NanoVNA Saver add additional Thrurefl calibration step, this allow additional compensate error)

DiSlord, 2024-06-12:
Test this code
https://groups.io/g/nanovna-beta-test/topic/nanovna_h4_ethanced_responce/10663625

owenduffy, 2024-06-12:
Thank you, I installed NanoVNA.H4.v1.2.32.bin.

[image]

It made no significant difference to my test case of 200R @ 1MHz. Note above that it shows s11 reflection Z as 249.8-j0.194 which is expected, but R21ser=227.3 and r21xser=0.048.

BTW, did you fix the phase inversion of raw s21?

Owen

owenduffy, 2024-06-12:
I did another test on a quite small ferrite cored inductor. @10Mhz s11Z=2960+j2060.

 
[image]

So, this shows 4000+j1870.

Owen

DiSlord, 2024-06-12:
> Thank you, I installed NanoVNA.H4.v1.2.32.bin.
you not enable Enhanced responce on calibration page

owenduffy, 2024-06-12:
> you not enable Enhanced responce on calibration page
> Enable it

Ah... yes, that makes a difference.

BTW the spelling is RESPONSE.
[image]

That looks much better, thanks again.

Owen

DiSlord, 2024-06-12:
Yes, I fix this later (already in 1.2.32 firmware, just old screenshot)

DiSlord, 2024-06-13:
Added in https://github.com/DiSlord/NanoVNA-D/commit/ace0512cf605d0dd94ae5ee9ff35a4db11802ab8
Port2 compensation not appy, need think how free memory for e22 calibration data (Port2 impedance)

owenduffy, 2024-06-13:
You may be interested in my changes to NanoVNA-App to do Enhanced Response correction: https://github.com/owenduffy/NanoVNA-App/blame/main/Calibration.cpp .

owenduffy, 2024-06-13:
It appears that v1.2.32 is a good fix for this issue.

Owen

DiSlord, 2024-06-14:
I update NanoVNA-App, add full ER calibration include port2 impedance compensation
https://github.com/DiSlord/NanoVNA-App/commit/6836d130576f98cc4528cc79396e94e462dd61f8

closed 2024-06-14 by DiSlord