DiSlord/NanoVNA-D #84: Enhancement: one touch PAUSE/RESUME and one touch SCREENSHOT

owenduffy, 2024-06-19:
It would be very convenient if one could PAUSE / RESUME by touching an icon on the screen.

It would be very convenient if one could take a SCREENSHOT forcing AUTONAME for the duration and automatic RESUME by touching an icon on the screen.

[image]

The above screenshot has had a P and S added at lower right as a suggested interface to these two functions. You could change the colour of the P to show paused or not, and change the colour of the S to show 'working'.

Owen

DiSlord, 2024-06-19:
If click on BW text at bottom, must run screenshot procedure
If auto name enabled - auto or ask name

owenduffy, 2024-06-19:
I thought I checked by touching the left margin of the screen that it was unused, but I see it is now used for adjusting scale... though in a little exploration I kept getting into a state where I needed to use the menu to straighten things out.

I did not know of the BW trick.

When I take a screenshot, I usually PAUSE so that I am sure the screenshot is going to have what I want to show, then take the screenshot and if necessary RESUME scan. It is a lot of keystrokes, and for a common operation I thought was I proposed would be very convenient.

Owen

DiSlord, 2024-06-20:
In LiteVNA then user enter filename sweep pause and resume after, I also add this on Nano

DiSlord, 2024-06-20:
Add pause sweep on screenshot (include filename input)
https://github.com/DiSlord/NanoVNA-D/commit/413dd210de5a347fa8d27df0e6dffb3bea384c57

closed 2024-06-20 by DiSlord