DiSlord/NanoVNA-D #85: Enhancement: allow specification of DC resistance of calibration LOAD

owenduffy, 2024-06-19:
As I understand it, the calibration / correction process in NanoVNA assumes the LOAD is exactly 50+j0 ohms.

Inexpensive calibration parts are available with an accurately measured DC resistance (eg from SDR-KITS). Alternatively Rdc can be measured with a high resolution 4W DMM.

Taking that DC resistance into account when calibrating can greatly improve the calibration accuracy, specifically the Port 1 Directivity.

It would be really useful to have the option of specifying the DC resistance to say 3 decimal places, and to use that value to calculate the error terms.

This ought to be a fairly simple change with very useful benefit.

Thanks
Owen

owenduffy, 2024-06-19:
To give an example, the loads I received with two NanoVNA from Zeenko store measure 50.949 and 51.273 ohms. That essentially gives the NanoVNA Directivity of 40.5dB and 38.0dB, so it has excessive uncertainty for |s11| measurements below -29dB and -27dB respectively. This would be improved greatly by taking the DC resistance into account when calibrating.

Owen

DiSlord, 2024-06-20:
This feature (calibration standard correction) implemented in LiteVNA or possible use NanoVNA-Saver (not implemented offset loss and offset Z correction for Short/Open/Load)

LiteVNA have more powerful MCU and allow in real time calculate correction:
Calibration standard model LOAD must have Offset loss, Offset Z and Offset delay settings + R, L and C (if need work on high frequency range) 

More info in this thread:
https://groups.io/g/nanovna-beta-test/topic/calibration_standard/104889410

Possible add in H4, but calculation very expensive in calculations (lot of complex data)

DiSlord, 2024-06-20:
Here is NanoVNA saver code for calibration standard correction (additional need calculate open/short/load impedance depend from parameters like offset loss/ offset impedance/ offset delay, parasite L or C, and R for Load):
  complexf gm1 = data[CAL_SHORT];  // g1 real short impedance
  complexf gm2 = data[CAL_OPEN];   // g2 real open impedance
  complexf gm3 = data[CAL_LOAD];   // gl real load impedance
// NanoVNA Saver version (later use g1` = 1/g1 and g2` = 1/g2 values!!!)
  complexf denominator = (g1 * (g2 - g3) * gm1 + g2 * g3 * gm2 - g2 * g3 * gm3 - (g2 * gm2 - g3 * gm3) * g1);
  complexf e00 = - ((g2 * gm3 - g3 * gm3) * g1 * gm2 - (g2 * g3 * gm2 - g2 * g3 * gm3 - (g3 * gm2 - g2 * gm3) * g1) * gm1);
  complexf e11 = ((g2 - g3) * gm1 - g1 * (gm2 - gm3) + g3 * gm2 - g2 * gm3);
  complexf delta_e = - ((g1 * (gm2 - gm3) - g2 * gm2 + g3 * gm3) * gm1 + (g2 * gm3 - g3 * gm3) * gm2);
  return (dut * denominator - e00) / (dut * e11 - delta_e);

All data complex, as can see calculation very expensive, H not have hardware FPU, H4 have but add this part not easy task.

owenduffy, 2024-06-20:
Hi,

Thanks for thinking about this.

Yes, I see the calcs involved.

I am not suggesting a polynomial for ANY of the SOL parts, no provision for parasitic reactances or delays. Just:
gm1=-1+j0
gm2==1+j0
gm3=(R-50)/(R+50)+j0  where R is the user specified Rdc of the LOAD.

Some of the error term formula you give can be simplified because gm1 and gm2 are simple  + / - multipliers, and gm3 is a real float.

Noted that the -H would be slower due to hardware, I suspect more people buy -H4 than -H these days for lots of reasons, and this would be another reason.

So, if the code is too large or too slow on -H, but is practical on -H4, then it would be a very useful improvement on -H4.

Your thoughts?

Owen

DiSlord, 2024-06-21:
I check, for this possible use Port Z -> XX option (feature allow get results as measure XX system)
Example if need measure 75 om system, calibrate Nano by 50 Om calibration kit, and made Port Z -> 75, and you get results for 75 om system

Anyway if you calibrate by XX load, and need get result as on 50 om set Port Z -> 2500 / XX

owenduffy, 2024-06-21:
Thanks.

I tried to find that menu item... and I know I have seen it, but I cannot find it in v1.2.33.

Can you give me the full menu path pls?

Owen

DiSlord, 2024-06-21:
Display->Port Z

owenduffy, 2024-06-21:
[image]

Not on my -H4 v1.2.33

Owen

DiSlord, 2024-06-21:
You build from sources?
Port-Z - not included in open source part
Use this compiled firmware
https://groups.io/g/nanovna-beta-test/message/4708

owenduffy, 2024-06-21:
I installed that right now, and no change.

Owen

DiSlord, 2024-06-21:
Hmm, I download firmware from post, erase firmware on device and upload new (from post)
[image]
[NanoVNA H4 v1.2.33.zip](https://github.com/user-attachments/files/15933612/NanoVNA.H4.v1.2.33.zip)

owenduffy, 2024-06-21:
Ok, I have a script to load the firmware... and the blanks in the filename tricked the script.

Got it loaded now, I will try it.

Owen

owenduffy, 2024-06-21:
OK, so the VNA was calibrated with a LOAD very close to 50 ohms.

I am now measuring a termination with R about 51.5 ohms.

[image]

I set P>Z to 51.47 and it changes the calculated |s11|, but it does not change Z reported in the Smith chart result, it does appear to have move the Smith chart plot... it is a very small move and would be more apparent if I try a larger change.

Owen

owenduffy, 2024-06-21:
So, with 25 ohms:

[image]

The Smith chart plot looks good, the |s11| looks good, the Z written against the Smith trace marker is not adjusted.

DiSlord, 2024-06-21:
In this mode all Impedances data on markers recalculated to Port-Z system Z0 = Port Z

DiSlord, 2024-06-21:
Ok, now I know how add easy correction for Load real R, try add this later for tests

PS more easy check this in NanoVNA Saver, need use 
[image]
Load->Resistance set to your value

owenduffy, 2024-06-21:
Ok, so I recalibrated and used a LOAD approx 51.5.

I then measure a termination VERY close to 50. (49.805)

[image]

|s11| is real low @ 1MHz... good.
But Smith marker data shows R=48.6... and it is ACTUALLY 49.805.

So, this does not properly correct a bad calibration LOAD.

Owen

PS: I do not use saver, I gave up downloading versions that crashed.

owenduffy, 2024-06-21:
BTW, I am not suggesting the displayed impedance is wrong. It is consistent with the -80dB |s11|.

My point is that this P>Z adjustment has an application, eg if you wanted to make measurements wrt Zo 75 ohms, but it is not a way to correct for a calibration LOAD error.

Owen

DiSlord, 2024-06-21:
Yes, as I write, On enable Port Z, impedances in markers recalculated to target Z0 = Port Z value, need disable this for your task
I try add this, need check formulas

owenduffy, 2024-06-21:
As I see it, a s11->Z->s11 can adjust the s11 for the LOAD error, but the recalculated s11 needs to be interpreted in the 50 ohms context for the case we are discussing of correcting calibration with a 51 ohm LOAD.

That is different to the way P>Z current works, and is intended to work.

Owen

DiSlord, 2024-06-22:
Added calibration standard LOAD R compensation (only this)
[image]
In CALIBRATE menu now possible input real LOAD calibration standard R value
S parameters compensation calculated from Port Z and calibration standard LOAD R
Port Z also work, marker impedance calculated to target Z0 = Port Z

[nanoVNA H4 v1.2.35.zip](https://github.com/user-attachments/files/15935895/nanoVNA.H4.v1.2.35.zip)

If all ok I upload this

owenduffy, 2024-06-22:
Thanks, got it, I will test it get back within 20h.

Owen

owenduffy, 2024-06-22:
Ok,

I calibrated with a LOAD that measured 51.5 ohms.

[image]

Now measuring a pretty good LOAD (49.805) I get the above.

That looks good, I will do some more testing.

Thanks
Owen

owenduffy, 2024-06-22:
HI,

Let me call the new parameter Rcal for this message.

As I understand it, the process path now is raw measured s11 is corrected using the saved calibration dataset, then if Rcal!=50, you transform corrected s11 to Z using Rcal then transform Z to s11 using 50.

All this is done before any other use of s11 (eg in source match compensation of s11), so everything uses the corrected and Rcal adjusted s11, everything is consistent with that?

Please confirm.

I see you save Rcal as part of the calibration slot... good, that is what is needed.

Thanks, I think this is another really useful improvement.

Owen.

DiSlord, 2024-06-22:
I write before full calibration standard correction formulas:
complexf gm1 = data[CAL_SHORT]; // g1 real short impedance
complexf gm2 = data[CAL_OPEN]; // g2 real open impedance
complexf gm3 = data[CAL_LOAD]; // gl real load impedance
complexf denominator = (g1 * (g2 - g3) * gm1 + g2 * g3 * gm2 - g2 * g3 * gm3 - (g2 * gm2 - g3 * gm3) * g1);
complexf e00 = - ((g2 * gm3 - g3 * gm3) * g1 * gm2 - (g2 * g3 * gm2 - g2 * g3 * gm3 - (g3 * gm2 - g2 * gm3) * g1) * gm1);
complexf e11 = ((g2 - g3) * gm1 - g1 * (gm2 - gm3) + g3 * gm2 - g2 * gm3);
complexf delta_e = - ((g1 * (gm2 - gm3) - g2 * gm2 + g3 * gm3) * gm1 + (g2 * gm3 - g3 * gm3) * gm2);
S11 = (dut * denominator - e00) / (dut * e11 - delta_e);

gm1 = -1
gm2 = 1
gm3 = 0 before, but now calculated from Port-Z and Load R values
gm3 = ("Load R" - "Port-Z")/ ("Load R" + "Port-Z")

Delays and loss not apply, also used close source optimizations'

Z parameters calculations from complex S parameters (z0 - target impedance = Port Z)
Z = z0 * (1 + S) / (1 - S) = R + jX

owenduffy, 2024-06-22:
Thanks.

As far as I have tested it (and that is a few hours of testing), it looks good and is a very worthwhile improvement.

[Dislord’s NanoVNA-D firmware v 1.2.35 includes a facility to apply a correction based on the DC resistance of the LOAD](https://owenduffy.net/blog/?p=33038) contains a description of some of the tests,

Well done.

Owen

closed 2024-06-23 by DiSlord

owenduffy, 2024-08-05:
> Port-Z - not included in open source part

Why is that?

Owen