DiSlord/NanoVNA-D #88: Missing prompt after pause command when using uart interface

owenduffy, 2024-07-09:
It appears that sometimes, the VNA does not issue a ch> prompt following a pause command on the uart interface at 38400bps in this example.

Here is a log extract from NanoVNA-App around the time I request a single sweep from NanoVNA-App:

```
14.998 rx: ch> 
14.998 tx: marker
15.012 rx: marker
15.012 rx: 1 160 41000000
15.012 rx: ch> 
15.012 tx: sweep
15.175 rx: sweep
15.175 rx: 1000000 101000000 401
15.185 rx: ch> 
15.187 tx: pause
15.187 rx: pause
27.154 rx: timeout pause lines: 0 data: 0/0
27.157 tx: pause
27.170 rx: ch> 
27.170 tx: bandwidth 3
27.178 rx: bandwidth 3
27.186 rx: bandwidth 3 (1000Hz)
27.188 rx: ch> 
27.188 tx: power 255
27.217 rx: power 255
27.219 rx: ch> 
27.223 tx: scan_bin 1000000 101000000 401 7
27.281 rx: scan_bin 1000000 101000000 401 7
30.585 rx: 8024 bytes received
30.588 rx: ch> 
30.593 tx: resume
30.648 rx: resume
30.651 rx: ch> 
```

Note that at timestamp 15.187, a ch> prompt is not issued and about 12s later, NanoVNA-App times out and reissues the pause command which works this time.

Here is the same transaction using USB interface:

```
44.146 tx: marker
44.174 rx: marker
44.174 rx: 1 160 41000000
44.185 rx: ch> 
44.190 tx: sweep
44.190 rx: sweep
44.190 rx: 1000000 101000000 401
44.206 rx: ch> 
48.642 tx: vbat
48.643 rx: vbat
48.644 rx: 3968 mV
48.644 rx: ch> 
48.644 tx: marker
48.646 rx: marker
48.647 rx: 1 160 41000000
48.648 rx: ch> 
48.648 tx: sweep
48.674 rx: sweep
48.675 rx: 1000000 101000000 401
48.675 rx: ch> 
48.675 tx: pause
48.738 rx: pause
48.739 rx: ch> 
48.739 tx: bandwidth 3
48.739 rx: bandwidth 3
48.739 rx: bandwidth 3 (1000Hz)
48.755 rx: ch> 
48.758 tx: power 255
48.758 rx: power 255
48.766 rx: ch> 
48.770 tx: scan_bin 1000000 101000000 401 7
48.801 rx: scan_bin 1000000 101000000 401 7
50.001 rx: 8024 bytes received
50.011 rx: ch> 
50.065 tx: resume
50.066 rx: resume
50.070 rx: ch> 
```

I have never seen this happen on the USB interface, but it happens frequently on the uart interface. I do not know if it affects other than the pause command. The equivalent pause command is at 48.675 and it receives an almost echo and ch> prompt.

It is as if having echoed the pause command text, the VNA does not respond further, perhaps it stops, times out, and restarts to accept the pause command sent about 12s later.

Owen

DiSlord, 2024-07-11:
Fixed in https://github.com/DiSlord/NanoVNA-D/commit/d9872b2cd994b1fd22e06df4ecd980f4cfebd439

closed 2024-07-11 by DiSlord

owenduffy, 2024-07-11:
Do you have a .bin to test?

DiSlord, 2024-07-12:
[H4.zip](https://github.com/user-attachments/files/16188611/H4.zip)
Unpack and flash to H4

owenduffy, 2024-07-12:
Thanks, I will report back within 24h.

owenduffy, 2024-07-12:
It works, it it no longer stalls on the pause command pending some timeout.

I tested it with a wire connection at 38400 and a HC-05 bluetooth adapter again at 38400 serial speed.

It worked fine with NanoVNA-App ordering up scans, singly or continuous scan, 401 points. I did test it on 1601 points... it appears to segment the scan so there is not more than 8k in each data transfer, no overruns observed.

I tried a screen dump with a .py script, and it crashed. It appears to transfer data, but not enough data for the screen size.

My first thought is there is no flow control, and the NanoVNA overruns the buffer in the bluetooth link.

It would be helpful if data streamed to the UART port was paced so it did not require a huge buffer... or use some form of flow control.

Thanks, fixing the hang on pause is really useful. It would be nice to screendump across the serial connection.

Owen

owenduffy, 2024-07-12:
Sorted the screen dump problem, it was a short timeout set in the receiving py script. Works fine with appropriate timeout.

So, it appears all good with current testing.

Thanks again.

Owen

DiSlord, 2024-07-13:
I write small notes for BT to NanoVNA connection
https://groups.io/g/nanovna-beta-test/topic/bluetooth_connection_for/107196889