DiSlord/NanoVNA-D #90: How to Achieve Successful Serial Communication Between Arduino and NanoVNA

orhanapaydin, 2024-07-25:
Hello,

I am working on a project where I need to establish a reliable serial communication between an Arduino and a NanoVNA. Specifically, I am using the NanoVNA H4 model and attempting to connect it via UART0 at a baud rate of 115200 to obtain S21 measurements and display them on a screen. I have a few questions regarding the connection options available under the Connection menu:

What is the exact purpose of the USB and serial connection options?
How can I successfully establish a serial connection with the NanoVNA using an Arduino?
Any guidance, sample code, or documentation that could help me achieve this would be greatly appreciated.

DiSlord, 2024-07-25:
You mean this menu?
[image]
Connection define command source USB or SERIAL
SERIAL SPEED - USART serial speed 
All options stored in config, so use save config after change

Send any shell command, receive answer, parse answer
You can use scan command:
scan {start(Hz)} {stop(Hz)} [points] [outmask]

Outmask:
#define SCAN_MASK_OUT_FREQ       0b00000001
#define SCAN_MASK_OUT_DATA0      0b00000010
#define SCAN_MASK_OUT_DATA1      0b00000100
#define SCAN_MASK_NO_CALIBRATION 0b00001000
#define SCAN_MASK_NO_EDELAY      0b00010000
#define SCAN_MASK_NO_S21OFFS     0b00100000
#define SCAN_MASK_BINARY         0b10000000

0b101 - outmask (receive frequency and S21 data)

scan 10000 20000 101 0b101
Answer example:
10.362 tx: scan 10000 20000 101 0b101
10.365 rx: scan 10000 20000 101 0b101
10.546 rx: 10000 0.000022667 -0.000046399
10.548 rx: 10100 -0.000056518 -0.000155888

orhanapaydin, 2024-07-25:
I have configured the NanoVNA to use SERIAL mode with a baud rate of 115200 and have created the following code to send commands and display the data on a TFT screen:

#include <TFT_eSPI.h>       // TFT screen library

// Screen and UART settings
#define SCREEN_WIDTH 320
#define SCREEN_HEIGHT 240

// TFT screen and UART objects
TFT_eSPI tft = TFT_eSPI(); // TFT screen object

void send_command(const String& cmd) {
    Serial.write(cmd.c_str()); // Send the command through the main serial port
    delay(50); // This delay is adjustable to allow for empty lines to pass
}

void display_message(const String& message) {
    tft.setCursor(10, 10);
    tft.fillScreen(TFT_BLACK); // Clear the screen
    tft.setTextColor(TFT_WHITE, TFT_BLACK); // Set text color to white with black background
    tft.setTextSize(1); // Set text size
    tft.println(message); // Print the message to the screen
}

void setup() {
    Serial.begin(115200); 
    delay(50); // Allow time for the serial connection to stabilize
    
    tft.init();
    tft.setRotation(1);
    tft.fillScreen(TFT_BLACK); // Clear the screen to black
    tft.setTextColor(TFT_WHITE, TFT_BLACK); // Set text color to white with black background
    tft.setTextSize(1); // Set text size
    tft.println("Starting up..."); // Print startup message to the screen
    delay(2000); // Wait for 2 seconds
    tft.fillScreen(TFT_BLACK); // Clear the screen to black
}

void loop() {
    send_command("scan 10000 20000 51 0b00000001");    
    String message = "";
    while (Serial.available() > 0) {
        char c = Serial.read(); // Read incoming character
        message += c; // Append character to the message
        delay(10); // Wait to ensure all data is received
    }
    if (message.length() > 0) {
        display_message(message);
    }
    delay(2000); // Slow down the loop a bit
}
------------------------------------------------------------- end code section

However, the TFT display does not show any frequency data. The display_message function should clear the screen and print the incoming serial data, but nothing appears on the display. Are there any specific things I should check or adjust to ensure the data is properly received and displayed?

DiSlord, 2024-07-26:
Tx pin on NanoVNA connected to Arduino Rx pin
Rx pin on NanoVNA connected to Arduino Tx pin
Gnd to Gnd
Correct?

Need add carriage return at end command
send_command("scan 10000 20000 51 0b00000001\r\n");

This command return only frequency list

orhanapaydin, 2024-07-26:
[image]

I connected the NanoVNA to esp32 like that. I use the USB cable to data communication. 

Fixed my code according to your suggestion ("\r\n"), but result is same.

DiSlord, 2024-07-26:
You connect NanoVNA as host USB?
NanoVNA not allow USBH mode, only USBD (usb device)

I also don't understand where you are sending the request, ESP32 acts as a USB bus controller and accesses NanoVNA as a USB CDC device?

orhanapaydin, 2024-07-26:
I connected the NanoVNA directly to the ESP32 via USB, powered the ESP32 through the circuit using an external 5V source, and ran the setup. The code I wrote at the top sends requests from the ESP32 to the NanoVNA. My expectation is for the NanoVNA to respond as requested and for the ESP32 to display the received response on the screen.

DiSlord, 2024-07-26:
I got it, and I assume you're doing something wrong.
NanoVNA is a USB CDC device

Connect it to your computer, use any terminal program (Putty for example)? open NanoVNA COM port, send a request and you'll get a response.

I don't see anything like that in your code

The Serial library is used, I'll just assume that your device is not a controller in this mode, it works the same way as NanoVNA expects someone to control it (work in USBD mode, not host)

orhanapaydin, 2024-07-27:
I understand. I will try to control the NanoVNA with another device like Arduino (connect Tx pin to Tx, Rx pin to Rx, GNG to GNG). I will share my results.

Ho-Ro, 2024-07-27:
> connect Tx pin to Tx, Rx pin to Rx

NOPE! `Tx` -> `Rx` and `Rx` <- `Tx`

Just understand what these names mean: `Tx` (Transmit) is an *output* and `Rx` (Receive) is an *input*.

closed 2024-08-17 by DiSlord