DiSlord/NanoVNA-D #92: NanoVNA-H

TimBastress, 2024-08-14:
I'm trying to update the firmware in my NanoVNA-H to the latest shown here of v1.2.29 using the .dfu file and stm32CubeProgrammer software on Windows.

It seems to recongnize the connection and it reads it ok, and it seems to wirte the .dfu file ok as well but when I power off my vna and power it back on I just see a white screen.  I cant seem to get it to boot back into the VNA screen.

[image]

any help would be greatly appreciated!

Tim

DiSlord, 2024-08-15:
Check HW version on back
SM_ST text?

If yes need use from here
https://github.com/hugen79/NanoVNA-H/releases
SM_ST version for H

PS soon I add LCD support in this firmware

closed 2024-08-17 by DiSlord

TimBastress, 2024-08-17:
the HW version on the back says 3.6_MS

It had a graphical video before i tried to upgrade the firmware.   So I'm really confused as to why I cant get it to restart after the upgrade "seemingly successful"