DiSlord/NanoVNA-D #95: edelay display and recalled sxp file

owenduffy, 2024-12-01:
It seems to me that when an .sxp is loaded from the SD card, that the e-delay value shown on the display is not relevant, expecially if it is non-zero, and should be hidden whilst displaying a recalled file?

Owen

DiSlord, 2024-12-02:
Possible only reset edelay settings on load snp files

owenduffy, 2024-12-02:
In understand that you cannot obtain e-delay from the snp file.

But if the VNA had e-delay set to say 1nS and THEN you load a snp file, the display continues to show 1nS... it should be changed to display 0.

Owen

DiSlord, 2024-12-02:
When loading, I simply stop the sweep and replace the measured values ​​with the data from the file.
Device don`t know data source and display as before (most easy way display data)

I can reset data modification like delays, Port-Z on load file. This hide display this values

owenduffy, 2024-12-02:
I think the best thing is to set all those things that would alter transformation to (50 ohm) defaults.

That way, the screen shots do not contain inconsistent data.

Owen

DiSlord, 2024-12-24:
Add reset edelays settings in
https://github.com/DiSlord/NanoVNA-D/commit/676280eee0b7d4e1161b2d0da38c330dfd4b3975

closed 2024-12-24 by DiSlord

owenduffy, 2024-12-24:
Thanks... Owen.